team4909/2018-PowerUp #88: Joystick Sensitivity Mapping func in BionicJoystick

roshanr10, 2018-01-20:
## Description of Changes Requested
- [ ] Joystick Mapping func in BionicJoystick 

`public double getSensitiveAxis(BionicAxis axis, double deadzone, double sensitivity_coeff)`

https://www.desmos.com/calculator/bhmtf7gvdt

Should be a cubic function using `public double getThresholdAxis`.

## Feature Request Contact (for addt. questions)
@roshanr10

jaspercoughlin, 2018-01-26:
```
public double getSensitiveAxis(BionicAxis axis, double deadzone, double sensitivity){
		double axisValue = getThresholdAxis(axis, deadzone);

		return (1 - sensitivity) * axisValue + sensitivity * Math.pow(axisValue, 3);
	}
```

roshanr10, 2018-01-27:
Can you PR this branch in please?

PR #90 (2018-01-27): Joystick Sensitivity
## Description of Changes
Used cubic function for joystick sensitivity.
## Referenced Issue
Closes #88
closed 2018-01-27 by roshanr10