team4909/2018-PowerUp #92: Arduino LED Code

roshanr10, 2018-01-28:
## Description of Changes Requested
I'd like to be able to send the Arduino controller a set of RGB values to set the eight LED strips(assume they're 64 LEDs each, as the existing single strip) to from the RoboRio. The Arduino is configured as an I2C slave while the RoboRio is an I2C master; the Arduino side should be using the Wire library to parse the messages.

Although the implementation details are wide open for you to architect, one way to do this would be to send the string `0,0,0/255,0,0/0,255,0/0,0,255/255,255,255/0,0,0/0,0,0/0,0,0/0,0,0&` to the Arduino and parse the received data to set the `r,g,b` values to all of the eight LED strips. 

As for the ampersand character (&), that is so we can read all of the characters from the I2C buffer, and then exit the loop when we hit that, denoting the last character in the "message". The "/" would then denote the individual LED strips. The characters themselves are unimportant but are an easy way to identify the separation between the data.

## Feature Request Contact (for addt. questions)
@roshanr10

djwalker200, 2018-01-30:
Can you add me to this Roshan?

ashwinc12, 2018-01-31:
`void receiveEvent(int bytes)
{
	uint8_t colorArray[4];
	int8_t i = 0;
	while (0 != Wire.available() && i < 4)
	{
		colorArray[i++] = Wire.read();
	}
	// Sending to LED

	for (i = 0, i < 16, i++) {
		strip.setPixelColor(i, colorArray[1], colorArray[2], colorArray[3]);
	}
	strip.show();
}`

ashwinc12, 2018-01-31:
void receiveEvent(int bytes)
{
	uint8_t colorArray[4];
	int8_t i = 0;
	while (0 != Wire.available() && i < 4)
	{
		colorArray[i++] = Wire.read();
	}
	// Sending to LED

	for (i = 0, i < 16, i++) {
		strip.setPixelColor(i, colorArray[1], colorArray[2], colorArray[3]);
	}
	strip.show();
}

roshanr10, 2018-01-31:
Is this code setting the same color to all the strips?

djwalker200, 2018-01-31:
void receiveEvent(int bytes)
{
  uint8_t colorArray[4];
                //creates Array of length 4. (strip, r , g, b)
  int8_t i = 0;
  while (0 != Wire.available() && i < 4)
{
    colorArray[i++] = Wire.read();
}
                        //sends Array components individually.
                        // Sending to LED

  for (i = 0; i < 16; i++) {
      strip.setPixelColor(i, colorArray[1], colorArray[2], colorArray[3]);
}
//sets pixel color to entire strip.

djwalker200, 2018-02-03:
package org.team4909.bionicframework.hardware.core;
import edu.wpi.first.wpilibj.DriverStation;
import edu.wpi.first.wpilibj.I2C;

import org.team4909.bionicframework.interfaces.Commandable;

/**
 * Arduino Library for I2C Communications to be used for LEDs and sensors
 */
public class Arduino {
	/**
	 * Enum of State Signals to send to an Arduino
	 */

	public boolean writeBulk(byte[] toSend)

	private I2C i2c;

	/**
	 * @param address I2C Address as configured on Arduino
	 */
	public Arduino(int address) {
		i2c = new I2C(I2C.Port.kOnboard, address);
	}

	/**
	 * @return Returns a Commandable that can be used by the operator and autonomous CommandGroups
	 */

	public void initialize() {
		byte[] toSend = {( byte)  };
		try {
			i2c.transaction(toSend, 1, null, 0);
		} catch (java.lang.NullPointerException e) {
			DriverStation.reportError("Could not connect to Arduino", false);
		}

		public void arduinoColors ( byte strip, byte r, byte g, byte b){

		}
		i2c.writeBulk(strip, 1);
		i2c.writeBulk(r, 2);
		i2c.writeBulk(g, 2);
		i2c.writeBulk(b, 2); //sends (strip,r,g,b)

	}

}

ashwinc12, 2018-02-03:
`#include <Math.h>
#include <Adafruit_NeoPixel.h>
#include <Wire.h>

#define PIXEL_COUNT 64
Adafruit_NeoPixel strip1 = Adafruit_NeoPixel(PIXEL_COUNT, 3, NEO_GRB + NEO_KHZ800);
Adafruit_NeoPixel strip2 = Adafruit_NeoPixel(PIXEL_COUNT, 4, NEO_GRB + NEO_KHZ800);

// Set Configuration
int lastValue = 0;
int deviceNumber = 4;

void setup() {
	Wire.begin(120);
	Wire.onReceive(receiveEvent);
	Serial.begin(9600);
}

byte x = 64;

void loop() {

	Wire.beginTransmission(8); // transmit to device #8
	Serial.print("x is" + x);
	Wire.write (x);   //sends 64 bytes
	Wire.endTransmission();

	x++;
	delay(500);
}

void receiveEvent(int bytes)
{
	uint8_t colorArray[4];
	int8_t i = 0;
	while (0 != Wire.available() && i < 4)
	{
		colorArray[i++] = Wire.read();
	}
	// Sending to LED

	for (i = 0, i < 16, i++) {
		strip.setPixelColor(i, colorArray[1], colorArray[2], colorArray[3]);
	}
	strip.show();
}`

roshanr10, 2018-02-03:
The code was revised today, should be able to wire up and test next meeting.

roshanr10, 2018-02-03:
For the time being, we can push this to the side; a single controllable LED strip(what we had going previously) should be enough to signify cube detection to the drivers, and all else is likely cosmetic... we can revisit this in another week if time allows.

All the cosmetic LEDs (like underglow and such) can likely just be set from the Arduino itself.

roshanr10, 2018-02-06:
Currently planning to revert to single controllable strip via I2C and running others manually; doing one strip for indication of control of cube and all others being done on the Arduino end for cosmetics.

closed 2018-02-06 by roshanr10