team4909/2018-PowerUp #96: LIDAR Lite v3 Research

roshanr10, 2018-02-03:
Given that we're aiming to be able to detect cubes without contact, we'd like to look into using the LIDAR Lite v3 from Garmin that we have. The module supports both I2C and PWM, we need to find which works best for us over the wire distance (~5-10ft) and is easier and more robust to program.

roshanr10, 2018-02-03:
So... I can't seem to find any evidence that the Rio has native PWM input, so that might make it more difficult... I'd look into I2C; we could go into the Arduino, but that's just more unneeded obscurity.

roshanr10, 2018-02-05:
Here's some relevant code samples from teams 401 and 3467... if you need a walkthrough of either of these examples, I can put you in touch with their software mentors; please let me know.

https://github.com/team401/2017-Robot-Code/blob/511f43203a2e8907310df10e172844a745c58f0e/src/main/java/org/team401/lib/Lidar.java

https://github.com/WHS-FRC-3467/Skip-5.5/blob/master/src/org/usfirst/frc3467/subsystems/LIDAR/LIDAR.java

sMyn42, 2018-02-09:
Would it hurt so much to run all of the Lidar functions throught the arduino, and send the results to the RoboRio?

roshanr10, 2018-02-09:
The RoboRio is connected to the Arduino via I2C anyway... the code should be very similar to read from the LiDAR or the Arduino. Thus it doesn't make sense to go through the Arduino and add more complexity.

sMyn42, 2018-02-09:
If you are picking this up for an Arduino, make sure you have a hefty capacitor on the power line, it pulls a fair bit of current on turn-on, they recommend a 680uf cap.

sMyn42, 2018-02-09:
Oh, never mind that last one, I didn't realize that u commented.

sMyn42, 2018-02-09:
We won't need to connect the yellow wire, obviously..

roshanr10, 2018-02-09:
Don't worry about the electrical part, Jared/Blake/Ben/myself can probably handle that; can you please focus on the code component that uses the LiDAR over the I2C bus as per their spec sheet and the above examples?

sMyn42, 2018-02-09:
I found an easy link that we could just edit, and the only complication would be to incorporate the arduino, which we've already done.

I'm currently unable to find what kind of data the lidar itself outputs, and how to read on the roboRio.

roshanr10, 2018-02-09:
This document should explain everything about the data that LiDAR exports, and we need to write our own I2C code for the reading values for the Rio.

http://static.garmin.com/pumac/LIDAR_Lite_v3_Operation_Manual_and_Technical_Specifications.pdf

Although the Arduino is another point of failure, that's not my primary concern. Reading values from the I2C bus is the same whether we get it from the Arduino or the LiDAR. You're writing virtually the same code anyway. Only, the former has it going through the Arduino instead of directly.

Please look through the above examples in more depth and I can help you more as need be.

closed 2018-02-10 by roshanr10