team4909/2018-PowerUp #99: Design Operator Interface

roshanr10, 2018-02-03:
Just a matter of tweaking the current `Robot.java` to make the controls more friendly to the driver/operator.

Ask @jaredbotte for more input on how to design the operator interface, and @roshanr10 if you've got additional questions about how to adjust the controls feeding into the subsystem.

jasonbishop11, 2018-02-07:
Controls spreadsheet - https://docs.google.com/spreadsheets/d/12tJWqa2bEwR5QXK4SlrEze-n4QdxQfEQYEtyWJtxAXo/edit?usp=drivesdk

roshanr10, 2018-02-07:
@jaredbotte what would you like for the dashboard (cameras/sensor readings/manual overrides) ?

roshanr10, 2018-02-07:
also could we explain more in-depth what climber "semi-auto" is, and why we need a separate cancel?

roshanr10, 2018-02-08:
I've pushed some changes to the branch `Feature-99` to allow for a toggle for the drivetrain shifter. Please pull updates from GitHub before making further changes.

closed 2018-02-10 by roshanr10