team4909/2018-PowerUp #108: Sensor Overrides

roshanr10, 2018-02-07:
We should have overrides for *ALL* of our sensor feedback, and try to make it feel natural otherwise...

Examples:
-Elevator encoder doesn't work, (brake mode kicks in to an extent, user can still operate)
-Elevator limit switches only stop motion if they're working (normally open)
-Stop intake when cube sensed, require manual release otherwise
-Stop outtake when cube not sensed, require manual release otherwise
-Drive encoders don't work, fallback to PercentOutput

This should be started by going thru what sensors we use and how we could implement a fallback gracefully. Please post the list here and we can then go through to finalize how we'll continue.

jasonbishop11, 2018-02-09:
System | Sensor | What could break
-- | -- | -- 
Drivetrain | Encoders | No motor data is recieved
Elevator | Lift encoders | No motor data is recieved
  | Limit switch | Elevator won't know when to stop going up
  | LIDAR | Intake/Outake won't know when to stop motors

roshanr10, 2018-02-10:
Elevator limit switch will fail gracefully on the SRX side, just won't rezero or stop driving into the floor.

LIDAR may not be used to stop motors for intake, no harm in not using for outtake. Should just be a boolean check if condition met.

Elevator PID should just not run PID to avoid driving down at max speed?

Drivetrain PID runs velocity mode, should just fallback to percent output.

closed 2018-02-16 by roshanr10

jasonbishop11, 2018-02-16:
[Dashboard.txt](https://github.com/FRCteam4909/BionicFramework/files/1732870/Dashboard.txt)